trezor/trezor-firmware #188: Binance serialization

ciny, 2019-05-28:
Based on discussion with @tsusanka I moved full message serialization (eg a message that can be posted directly to binance chain) to a separate branch until we decide what to do with it. Currently it supports serialization to binary marshall format. 

Work needed:
- refactor binary marshall serialization
- add support for amino json serialization.

Do we want to add this support?

matejcik, 2019-05-28:
is there any reason we would want to do that on Trezor, as opposed to host-side?

tsusanka, 2019-05-28:
Not really, I don't think we want to implement the complete serialization. This should be done by the wallet. Let's leave the branch, but unless needed let's not do it

prusnak, 2019-05-28:
I agree this shouldn't belong to the firmware. It might make sense to add it to the trezorctl though.

matejcik, 2020-02-24:
I believe there is no point in keeping this issue open. If we ever need to do local Binance serialization in the future, we can reopen.

closed 2020-02-24 by matejcik