trezor/trezor-firmware #194: Measure and plot firmware memory usage

jpochyla, 2019-05-28:
The test runner should keep track of peak memory needed in the run. We should be able to compare it between commits/branches etc.

- In debug unix build, and if environment variable `TREZOR_LOG_MEMORY` is set to non-zero, memory info containing the peak memory usage is printed out.

matejcik, 2021-11-01:
This should be very workable now that we have workflow-restarts, AND we might be able to hack something for the hardware tests as well.

Idea sketch: a debug message that is sent after every test? after every workflow? The firmware keeps track of (a) current peak memory usage, (b) number of GC collection cycles?
Both values are reset when sent to the host.

We can start with something easier for the emulator:
1. figure out a good dump point, "just before workflow restart" sounds OK
2. dump peak values into a file, along with some identifying information (workflow name? list of loaded modules?)
3. create a report after the test suite is done

hiviah, 2021-11-09:
After some experimenting, emulator is mostly ready for this as it tracks peak memory built-in. Measuring after workflow seems to be reasonable choice.

On device, there doesn't seem to be `MICROPY_MEM_STATS` available, so we have to hack peak measuring in there somehow. One possibility is to instrument `gc_collect` and record max memory across last gc sweeps (at the end of `gc_collect` to see how much live objects take). Then we'd reset this on every printout, so the next workflow is not affected.

closed 2023-10-23 by jpochyla