trezor/trezor-firmware #206: Implement FIDO2 credential selection

andrewkozlik, 2019-05-31:
In FIDO2, the relying party may provide multiple valid credentials. When this happens we automatically select the first one. The correct behavior is to display all of them to the user, using their friendly name along with other stored account information, and ask the user to select the one they want to use.
See https://fidoalliance.org/specs/fido-v2.0-id-20180227/fido-client-to-authenticator-protocol-v2.0-id-20180227.html#authenticatorGetAssertion

andrewkozlik, 2019-05-31:
Related to https://github.com/trezor/trezor-firmware/issues/205.

andrewkozlik, 2019-06-07:
We would like to have some kind of selector widget in BIP-0039 and SLIP-0039 recovery as well. If the user enters one of the mnemonic shares incorrectly, then we can show them the list of words they entered, have them select the one, where they made a mistake, and allow them to make a correction. 

So maybe we can use the same selector widget for SLIP-0039, FIDO2 and maybe even for selecting HOTP credentials from a list, see https://github.com/trezor/trezor-firmware/issues/33.

prusnak, 2019-09-13:
Done

closed 2019-09-13 by prusnak