trezor/trezor-firmware #219: Add DECENT (DCT) Support

Roznovjak, 2019-06-04:
Hi.
I would like to implement support for the DCT coin either into the official firmware or into our own version of the firmware.
There is no need to implement new cryptographic primitives. All necessary crypto-related code is already implemented in the TREZOR firmware.
Being a graphene-based blockchain, the scale of the project is similar to the EOS app (or probably even smaller). 
I see that you do not accept new coins if they are not in the market cap top 30. However, can I at least get some sort of support or guidance from you?
Project: https://docs.decent.ch
Explorer: https://explorer.decent.ch

tsusanka, 2019-06-05:
As you've mentioned we are currently not accepting coins not in the top 30. Feel free to fork this repository and work on your coin there and release your own firmware. If you need help you can find us on [gitter](https://gitter.im/trezor/community). Also, another coins PRs are a great source of help, see [EOS](https://github.com/trezor/trezor-firmware/pull/87) and [Ontology](https://github.com/trezor/trezor-firmware/pull/177) for example.

closed 2019-06-05 by tsusanka

matejcik, 2019-06-05:
just to note: I've been looking into Graphene a little, and it *might* be possible to extend our EOS impl to support other Graphene networks. I can't be sure about it though. Notably what's missing is a good description of tx serialization formats.
 But if the Transfer action was the same in all Graphene networks, we could extend EosSignTx to support it

Roznovjak, 2019-06-11:
Recently, I have tried to spot the differences and similarities between EOS, Steem, BitShares and DECENT blockchain. The results are as follows:
Fees: Steem and EOS do not use transaction fees. DECENT and BitShares use additional fee parameter in operations/actions.
TXs: Only EOS TXs contain _max_net_usage_words_, _max_cpu_usage_ms_ and _delay_sec_ params. 
Operations/actions: Different serialization: EOS knows how to serialize actions from the smart contract name. DECENT, Steem and BitShares prepend uint in serialization to identify the operation/action type.
Accounts: DECENT and BitShares use database ids to represent accounts, EOS and Steem use account names instead.
Assets: that's the most inconsistent part. Only DECENT and BitShares share the same interpretation and data types.
Serialization of basic data types should be the same on the networks.
From my observations, I assume it's not possible to create general implementation which can support all graphene networks. The _EosSignTx_ message would need to contain network type specifier and take appropriate actions based on that flag.

matejcik, 2019-06-11:
> From my observations, I assume it's not possible to create general implementation which can support all graphene networks. The _EosSignTx_ message would need to contain network type specifier and take appropriate actions based on that flag.

Bitcoin-like coins do the same thing, so yes, that would be the way to go. But the serialization situation is unfortunate. I was hoping that Graphene uses a serialization format like Protobuf or CBOR or something, where the serialization is structured. As it stands, we *could* meaningfully reuse the app and the messages, but each Graphene network would need its own sub-implementation.

So that's a no-go for now.