trezor/trezor-firmware #221: Recovery: do not hard fail when an incorrect word was entered

tsusanka, 2019-06-05:
In recovery, if the checksum fails, we could improve the UX, so it does not fail completely. Instead let's offer users, which word was wrong, which they can then fix.

matejcik, 2019-06-06:
how do we know which word was wrong? :)

tsusanka, 2019-06-06:
There is an autocorrection code:

> Implementations should not implement correction beyond potentially suggesting to the user where in the mnemonic an error might be found, without suggesting the correction to make.

matejcik, 2019-06-06:
ooooh this should apply to SLIP-39 only

ZdenekSL, 2019-07-02:
We need the UX part.

ZdenekSL, 2019-07-02:
@tsusanka Please check if this is not duplicate.

tsusanka, 2019-07-03:
The best way to approach this is probably to inform the user that 14th word (e.g.) is most likely the wrong one and offer the user to fix it.

However, it's important to note, that many users probably won't make a mistake in the word, rather skip one or write some twice etc. which this won't solve.

tsusanka, 2019-07-09:
UX prefers not to do this unless there is a strong data background we should implement this.

Moving to backlog.

ZdenekSL, 2019-08-21:
Closed after discussion

closed 2019-08-21 by ZdenekSL