trezor/trezor-firmware #230: Can't run on Linux

jpfleury, 2019-06-05:
I'm on Xubuntu 18.10 64-bit. According to the steps on https://github.com/trezor/trezor-firmware/tree/master/python I only have to do the following:

```
pip3 install --upgrade setuptools
pip3 install trezor
```

After doing these steps, I have trezor installed:

```
$ pip3 list --format=columns | grep trezor
trezor                       0.11.3    
```

but it won't run:

```
$ trezorctl version
Traceback (most recent call last):
  File "/usr/local/bin/trezorctl", line 31, in <module>
    from trezorlib import (
ImportError: cannot import name 'ontology'
```

matejcik, 2019-06-06:
AFAICT your `trezorctl` binary is out of sync with the trezor library. Maybe you installed python-trezor before with `sudo pip3 install trezor`? in that case you would need to `sudo pip3 install --upgrade trezor`.

alternately, do different combinations of `pip3 uninstall trezor` (with or without `sudo`, with or without `--user`) as long as it keeps uninstalling something, and then repeat `pip3 install trezor`

or you could simply delete `/usr/local/bin/trezorctl` and then try reinstalling

closed 2019-06-06 by matejcik

jpfleury, 2019-06-06:
I uninstalled all I could with and without sudo, and reinstalled setuptools and trezor, and it worked. Thanks a lot.