trezor/trezor-firmware #231: ValueError during 'make build_unix'

tuxxy, 2019-06-06:
When I run `make build_unix`, I get the following error:

```19:45:18 $> make build_unix
./tools/res_collect
processing file trezor/res/cancel.toig (115 bytes)
processing file trezor/res/pin_close.toig (275 bytes)
processing file trezor/res/confirm.toig (89 bytes)
processing file trezor/res/space.toig (34 bytes)
processing file trezor/res/lock.toig (72 bytes)
processing file trezor/res/swipe.toig (137 bytes)
processing file trezor/res/recovery.toig (147 bytes)
processing file trezor/res/header_icons/send.toig (104 bytes)
processing file trezor/res/header_icons/reset.toig (189 bytes)
processing file trezor/res/header_icons/nocopy.toig (147 bytes)
processing file trezor/res/header_icons/wipe.toig (163 bytes)
processing file trezor/res/header_icons/receive.toig (109 bytes)
processing file trezor/res/header_icons/wrong.toig (182 bytes)
processing file trezor/res/header_icons/recovery.toig (147 bytes)
processing file trezor/res/header_icons/cog.toig (152 bytes)
processing file trezor/res/check.toig (181 bytes)
processing file trezor/res/click.toig (139 bytes)
processing file trezor/res/left.toig (73 bytes)
processing file apps/management/res/small-arrow.toig (195 bytes)
processing file apps/webauthn/res/icon_bitfinex.toif (1322 bytes)
processing file apps/webauthn/res/icon_lastpass.toif (763 bytes)
processing file apps/webauthn/res/icon_duo.toif (574 bytes)
processing file apps/webauthn/res/icon_gandi.toif (1367 bytes)
processing file apps/webauthn/res/icon_keeper.toif (1852 bytes)
processing file apps/webauthn/res/icon_stripe.toif (766 bytes)
processing file apps/webauthn/res/icon_dashlane.toif (2482 bytes)
processing file apps/webauthn/res/icon_fastmail.toif (1108 bytes)
processing file apps/webauthn/res/icon_slush_pool.toif (1931 bytes)
processing file apps/webauthn/res/icon_fedora.toif (1510 bytes)
processing file apps/webauthn/res/icon_google.toif (1260 bytes)
processing file apps/webauthn/res/icon_bitbucket.toif (931 bytes)
processing file apps/webauthn/res/icon_dropbox.toif (940 bytes)
processing file apps/webauthn/res/icon_webauthn.toif (1400 bytes)
processing file apps/webauthn/res/icon_github.toif (1947 bytes)
processing file apps/webauthn/res/icon_gitlab.toif (1117 bytes)
processing file apps/homescreen/res/bg.toif (9522 bytes)
continuing with trezor/res/resources.py, no changes detected
scons -Q -j 4 CFLAGS="-DGITREV=835d5df1" build/unix/micropython 
scons: *** [build/unix/embed/extmod/modtrezorconfig/modtrezorconfig.o] ValueError : unsupported pickle protocol: 4
make: *** [Makefile:106: build_unix] Error 2
```

Any idea what's up here?

tsusanka, 2019-06-06:
Use pipenv. `pipenv sync` and `pipenv run make build_unix`

prusnak, 2019-06-06:
Please reopen if that didn't help

closed 2019-06-06 by prusnak

tuxxy, 2019-06-06:
Of course it was that. :sweat_smile: 

Thanks for the eyes, folks. :)