trezor/trezor-firmware #232: SLIP-39 protocol for Trezor Wallet integration

tsusanka, 2019-06-06:
We should first design the "protocol" (what messages are exchanged) for the SLIP-39 integration in Wallet. Since almost everything is done on the device, it should be quite simple. So far, in the [tsusanka/slip39 branch](https://github.com/trezor/trezor-firmware/commit/f0bd96b369c53a114fd77454f235fccffe46b2a3#diff-b00d8fe315e1298939ff19a0ec1d3ddeR194) there is just a simple bool in ResetDevice.

- [ ] Is it all we need?
- [ ] Add ButtonRequest messages or similar for every screen on Trezor, so we can guide the user properly through the process.

This can be done in paralel with https://github.com/trezor/trezor-firmware/issues/59

prusnak, 2019-06-06:
We need also shamir flag in the RecoveryDevice, no?

prusnak, 2019-07-01:
I guess this is resolved now.

tsusanka, 2019-07-01:
Well to some extent. This is not yet done:

> Add ButtonRequest messages or similar for every screen on Trezor, so we can guide the user properly through the process.

But it's a minor issue I guess.

prusnak, 2019-07-01:
Reopening then

ZdenekSL, 2019-09-05:
Decided that we are going to do this.

tsusanka, 2019-09-05:
Seems like #35 is a prerequisite.

tsusanka, 2019-09-18:
Related to #530

ZdenekSL, 2019-09-26:
Closed after discussion

closed 2019-09-26 by ZdenekSL

prusnak, 2019-09-26:
We'll deal with that when solving bigger task at https://github.com/trezor/trezor-firmware/issues/35