trezor/trezor-firmware #235: Create a tool to upload and display static image

tsusanka, 2019-06-07:
Petr (UX designer) asked for a tool, which would upload a static image (like PNG) to trezor device, so he can have a look what it looks like at Trezor.

We have a converter from PNG in `tools/` and I can use `res.load` to show the image?

If that's the case, it should be pretty quick.

tsusanka, 2019-06-07:
@prusnak could you estimate how difficult this is? If it's a quick thing I think we might do it asap? So Petr can use it

prusnak, 2019-06-07:
Yes, I'll do it.

prusnak, 2019-06-09:
Added a quick hack to [toif-upload](https://github.com/trezor/trezor-firmware/tree/toif-upload) branch.

It modified ApplySettings.homescreen field to overwrite whatever is on the screen right now, instead of setting a home screen.

Also, there is a tool `core/tools/toif_upload` which takes 240x240 RGB (no alpha) PNG file as a parameter and uploads it to the device.

tsusanka, 2020-05-14:
I have rebased the toif-upload branch on top of master.

tsusanka, 2020-05-14:
@prusnak how difficult would it be have the same functionality for T1? Petr would really really make use of it :)

prusnak, 2020-05-14:
Relatively easy.

prusnak, 2020-05-23:
For T1 we can use a set homescreen feature. The homescreen can be set from the web wallet or using `trezorctl set homescreen -f filename`

closed 2020-05-23 by prusnak