trezor/trezor-firmware #236: Add display rotation to Features

tsusanka, 2019-06-07:
Let's return what rotation is used in Features. It might be useful for the Wallet settings to display the now selected value

prusnak, 2019-06-07:
I would like to see more reasoning why this is needed. Otherwise I see this just as a nice to have feature.

tsusanka, 2019-06-07:
Zorwy wanted to display, which setting is currently used. In other words highlight one of those buttons:

[image]

There is no other way to find out this, is there?

prusnak, 2019-06-07:
This adds unnecessary complexity. We don't show currently set homescreen in the homescreen selector either. Just show buttons without the highlight.

tsusanka, 2019-06-07:
Ok. Cc @TomasZorvan

closed 2019-06-07 by tsusanka