trezor/trezor-firmware #240: T One should remember passphrase

zack-shoylev, 2019-06-11:
T One device should remember last passphrase (or a list of passphrases) entered on device (as opposed to on host) until powered off.

It's too cumbersome to re-input the passphrase every time on device when needed. It makes me use the on-host input instead, which is less secure. Potentially related to https://github.com/trezor/trezor-firmware/issues/1

prusnak, 2019-06-11:
The device is designed that way. What you see is an issue with the web wallet and will be fixed soon.

closed 2019-06-11 by prusnak

zack-shoylev, 2019-06-12:
I think understand, @prusnak ; it seems host-side services should be caching the passphrase and re-using it as needed. However, the problem is that there might be multiple host-side services using the device: wallets, ssh agent, password manager, u2f logins. They don't seem to share, though. Is that something that would be fixed in the bridge? Curious since re-entering the passphrase on the device throughout the day is not too fun.

prusnak, 2019-06-12:
We are redesigning how passphrase will work in https://github.com/trezor/trezor-firmware/issues/1