trezor/trezor-firmware #244: Discussion: blinding key derivation for confidential transactions

romanz, 2019-06-13:
Following #66, we would like to allow TREZOR users to derive their blinding keys on the device.

In confidential transactions, the sender and the receiver use ECDH to derive a shared nonce, which is then used for hiding/recovering of the actual value and asset type being transacted.
In Elements/Liquid, the receiver uses the following derivation scheme for his ECDH public/private keys:
```
blinding_private_key := HMAC_SHA256(key=master_blinding_key, msg=script_pubkey)
blinding_public_key := secp256k1_publickey(private_key=blinding_private_key)
```
The receiver is using `blinding_public_key` construct a "blinded address", which is used by the sender to blind the relevant transaction outputs. Each such blinded transaction output also contains the sender's ECDH public key, so the receiver would be able to recover the shared nonce using its `blinding_private_key`.

An additional use-case is sharing some/all of the receiver's blinding private keys with an external auditor, allowing unblinding the audited outputs without being able to spend them.

## Derivation scheme
We would like to use similar blinding key derivation scheme on TREZOR and would like to propose the following derivation scheme for the `master_blinding_key`:
```
node := keychain.derive([HARDENED | 77])
master_blinding_key := sha256(node.private_key()).digest()
```
### Notes
* The specific derivation path is TBD - I have randomly chosen `m/77h` since "purpose=77" is not allocated yet in https://github.com/satoshilabs/slips - so it (hopefully) won't collide with other BIP-44 users.
* We use SHA256(...) of the derived private key, to allow the user share the resulting `master_blinding_key` with auditors, without exposing the underlying BIP-32 private key.

## Possible implementation
The suggestion above can be implemented by adding the following methods to the [Keychain class](https://github.com/trezor/trezor-firmware/blob/0b03794feec48514a58c2472bd5848c9f398da81/core/src/apps/common/seed.py#L10):
```python
def derive_blinding_private_key(self, script: bytes, curve_name: str) -> bytes:
    node = self.derive([HARDENED | 77], curve_name)
    derivation_key = hashlib.sha256(node.private_key()).digest()
    return hmac.new(key=derivation_key, msg=script, 
                    digestmod=hashlib.sha256).digest()

def derive_blinding_public_key(self, script: bytes, curve_name: str) -> bytes:
    private_key = self.derive_blinding_private_key(script, curve_name)
    return curve.secp256k1.publickey(private_key)
```

Please let us know what do you think about this proposal.

## Links
Some background about blinding/unblinding process can be found here:
* [An investigation into Confidential Transactions](https://github.com/AdamISZ/ConfidentialTransactionsDoc/blob/master/essayonCT.pdf) by @AdamISZ
* [Confidential Transactions tutorial](https://elementsproject.org/elements-code-tutorial/confidential-transactions#blindingkey)
* [Elements' blinding key derivation](https://github.com/ElementsProject/elements/blob/a6beb256ed5195c2a1014a34fdf354d5797247a8/src/wallet/wallet.cpp#L5594)
* [Elements' output unblinding using ECDH](https://github.com/ElementsProject/elements/blob/66c015529e7846f8491bcafd986326bcafc1bfcb/src/blind.cpp#L53)

prusnak, 2019-06-15:
So far there is no BIP77 allocated, so this should not collide with the __existing__ BIP. However there might be one in the future.

I'd suggest to change the 77h constant to 10077h. This way we can lock it in the SLIP77. (SLIP16 uses 10016h for example).

romanz, 2019-06-15:
Sounds good, may I prepare a SLIP draft?

prusnak, 2019-06-15:
Sure thing

andrewkozlik, 2019-06-15:
I see nothing wrong with your proposal @romanz, nevertheless I would like to ask you to hold off on preparing the SLIP. If I recall correctly, all you need is a symmetric key derived from the user's master seed. This is by far not the only place where something like that is needed, e.g. password encryption or, most recently, FIDO2. I would like us to reevaluate how symmetric keys are derived from the master seed. There are two reasons:

1. BIP-0032 was designed to derive public/private secp256k1 keys for hierarchical wallets, it doesn't make a lot of sense to use it for deriving symmetric keys and I am not sure what the security implications are.

2. The fact that BIP-0032 nodes are indexed by a 32-bit value is OK for hierarchical wallets, but there are instances where it would be more convenient to be able to specify the derived key using, say, a 256-bit value. I am thinking mainly FIDO2/U2F now.

I asked @onvej to do some research into this, find the BIPs and SLIPs, where symmetric keys are derived, think about the security implications and see if there are any other standards or recommendations for hierarchical key derivation. I think this is one of them: https://nvlpubs.nist.gov/nistpubs/Legacy/SP/nistspecialpublication800-108.pdf. I want to discuss this at our firmware security meeting on Wednesday. I don't know if it will have any implications for this issue, just giving a heads up.

romanz, 2019-06-16:
Good point, thanks for the heads up!

andrewkozlik, 2019-06-20:
I started writing a document which we might choose to adopt as a SLIP for the deterministic derivation of a hierarchy of symmetric keys. It's just a bunch of ideas, none of this has been approved yet. You are welcome to comment on it: 
https://docs.google.com/document/d/1QtwF8DF0TyPt9KFp5EsVpOY5d8PMicqafb2yGFoKfZ0/edit#

real-or-random, 2019-06-24:
@andrewkozlik All of these ideas look reasonable but with my cryptographer's hat on I'm strongly in favor of Proposal 3 (and then using HMAC-SHA512 to derive the master key, as you suggest):
 - It's similar to BIP32, which is nice because people know that already.
 - It will use only HMAC-SHA512, so no need for a second function, and we build on the assumption that HMAC-SHA512 is suitable for key derivation, which we accepted anyway.
 - As you point out, the ability to derive child nodes is independent of knowledge of the key, which gives a lot of flexibility for future applications but has no security drawbacks. (And Proposal 4, which achieves this property as well, is just slower for no real reason.)
 - Having 256 bits of entropy on each node is sufficient and perfectly fine for the future.

greenaddress, 2019-06-24:
@prusnak @andrewkozlik - thanks for all the feedback - I have been working on a software implementation of liquid for Blockstream Green and I am very keen on being compatible with Trezor for Liquid as we already are for Bitcoin. We are in the latest stages of testing and we'd like to make sure the compatibility works well.

In regards to vanilla BIP32 private keys derivation unfortunately we don't use BIP44 yet and currently understand that this will cause the device to prompt the user when using different BIP32 derivation schemes and we will have to live with this until we support BIP44 (w/ SLIP77 10077h path) but at least the device will be able to sign transaction correctly provided the user confirms.

For the master blinding key scheme the situation is a bit different as unless we are using exactly the same scheme the device won't be able to blind/unblind UTXOs. In this regards we currently use what's described in @romanz initial proposal but we would be happy to change to any agreed method provided we can commit to that. Our time frames are a bit compressed on this and we would be grateful if we could agree in principle (obviously provided no security issues are found with the proposal later)

In regards to preferred proposal I was original in favor on the @romanz as we had discussed and reviewed that however I am also in favor of Proposal 3 - this is just a preference, any other proposal seems also fine.

andrewkozlik, 2019-06-27:
Thank you for your feedback @real-or-random and @greenaddress. We selected Proposal 3 for the reasons that @real-or-random gave. The Hierarchical derivation of symmetric keys will be specified in SLIP-0021: https://github.com/satoshilabs/slips/pull/675.

prusnak, 2019-06-27:
I guess we can close this one.

closed 2019-06-27 by prusnak