trezor/trezor-firmware #245: Add support for FIDO2 resident credentials

andrewkozlik, 2019-06-13:
Supporting authenticator resident credentials makes it possible to select the credential private key given only an RP ID. This will allow Trezor to be used for first factor authentication (password-less) and it is also required for Microsoft compatibility (https://docs.microsoft.com/en-us/windows/security/identity-protection/hello-for-business/microsoft-compatible-security-key).

In order to support this feature we will need to be able to store the Credential IDs on the SD card https://github.com/trezor/trezor-firmware/issues/162.

andrewkozlik, 2019-08-09:
Passwordless login is already possible with resident credentials stored in NV RAM, which has implemented as part of https://github.com/trezor/trezor-firmware/pull/394.

prusnak, 2019-09-26:
Blocked by https://github.com/trezor/trezor-firmware/pull/323

vielhuber, 2019-11-04:
Thanks for this awesome work.

May I ask, what's the current status of this ticket?
Here is mentionned, that FIDO2 is already implemented in Trezor Model T, is that correct?
https://wiki.trezor.io/FIDO2
Will this be available also on the Trezor One?

andrewkozlik, 2019-11-05:
FIDO2 support is already fully implemented in Trezor T, including support for device resident credentials (i.e. passwordless logins) and the hmac-secret extension, which are the requirements for Microsoft compatibility. This ticket is only about the storage of the resident credentials on the SD card. Since we already allow storing up to 100 resident credentials in the Trezor T internal storage and we also support backing-up these resident credentials using `trezorctl`, the implementation of this ticket brings very little added value, making it very low priority.

AFAIK, there are no immediate plans for supporting FIDO2 on Trezor One. Correct me if I am wrong, @prusnak.