trezor/trezor-firmware #247: EOS unknown action handling is bad

matejcik, 2019-06-14:
as written (and apparently as designed), the flow for signing unknown actions is as follows:

consider a transaction with 3 actions: Transfer, Unknown, Transfer
(H)ost (T)rezor

H: `EosSignTx(num_actions=3)`
T: `EosActionRequest` (for action 1)
H: `EosActionTransfer(...)`
T: `EosActionRequest` (for action 2)
H `EosActionUnknown(data_size=123456, data_chunk=[7 bytes])`
T: `EosActionRequest(data_size=(123456-7))` (_still_ for action 2)
H: `EosActionUnknown(data_chunk=[555 bytes])`
T: `EosActionRequest(data_size(123456-7-555))` (still for action 2)
...
H: `EosActionUnknown(data_chunk=[remaining data])`
T: `EosActionRequest` (finally for action 3)

This way size of the data chunks is driven by host, not by Trezor. That's completely wrong, and unfortunately we didn't catch it before release. In addition, the number of EosActionRequests can't nicely correspond to the stated number of actions. And of course `data_size` in EosActionRequest is meaningless, it just tells "bytes remaining", which the host should know.

To keep backwards-compatible as much as possible, I propose the following changes:

1. add `action_index` field to `EosActionRequest`. That way the action requests are unambiguously paired to the respective actions
2. add `data_offset` field to `EosActionRequest`
3. enforce the following flow:
   T: `EosActionRequest(action_index=1)`
   H: `EosActionUnknown(data_size=123456, data_chunk=MUST be empty)`
   T: `EosActionRequest(action_index=1, data_offset=0, data_size=2048)`
   H: `EosActionUnknown(data_size=MUST be empty, data_chunk=[bytes 0-2047])`
   ...etc.

this accomplishes the following:
1. no backwards-incompatible protobuf change is needed, we're only adding fields
2. it's possible to have a host implementation that conforms to both: if you don't receive `data_offset`, ignore `data_size` and send next chunk as, say, 2048 next bytes
3. implementations for the new mode are much more sane

ccing @keepkeyjon, what do you think?

keepkeyjon, 2019-06-14:
In our implementation, `ActionUnknown` has a state machine that knows how much data is remaining, and decrements that each time we get a chunk of the action:

https://github.com/keepkey/keepkey-firmware/blob/97aef5cd5ff863998491ba34e1a35f364d688eee/lib/firmware/eos.c#L361

We check that the `data_size` of every message matches that of the first, and enforce that the amount remaining to be hashed is never less than the amount provided in the chunk. Finally, the confirmation happens when the amount remaining is `0`. One more check that we can add is to enforce that the `data_chunk.size` is always either filled, or equal the the size of the last remaining chunk.

I need to double check all of this for integer overflow/underflow, but I think it's solid. Is there a case I missed? Seems I need to review the trezor-core implementation again.

> add action_index field to EosActionRequest

I like that. Makes it easier to keep the host app and the confirm pages on device in sync, especially if we put those indices in the titles of each confirm page.

keepkeyjon, 2019-06-14:
> add action_index field to EosActionRequest

Thinking about this some more, should we also add that to EosActionCommon? I could see that bringing clarity to device logs, especially if enforced by the device that they never come out of sequence.

matejcik, 2019-06-14:
> We check that the `data_size` of every message matches that of the first

i see, we can easily keep that if needed

> and enforce that the amount remaining to be hashed is never less than the amount provided in the chunk. Finally, the confirmation happens when the amount remaining is `0`. One more check that we can [...]

that's sort of pointless, because `size` is prefilled by your protobuf library.

> I need to double check all of this for integer overflow/underflow, but I think it's solid. Is there a case I missed? Seems I need to review the trezor-core implementation again.

no, I think there are no security issues with your impl. (didn't go hunting for bugs though)
The problem is architectural: there's nothing stopping the host from just stuffing 1 MB of data into `data_chunk` -- and getting back an error because the device doesn't have enough memory. This goes contrary to the usual communication model, where the device _requests_ as much data as it can handle; here the responsibility is moved to the host. A wallet needs to know additional info, the maximum allowable size. (`messages-eos.options` is only needed for the C impl, it's not part of the protobuf definition. i see now that your max size is set to 256; we were planning to assume 2048)

matejcik, 2019-06-14:
> > add action_index field to EosActionRequest
> 
> Thinking about this some more, should we also add that to EosActionCommon? I could see that bringing clarity to device logs, especially if enforced by the device that they never come out of sequence.

We could do that - better in EosTxAck though, no? but i'm not sure if it does anything useful.
Either the host is truthful and sends the requested `action_index`, in which case it's not telling you much. Or it's doing something nefarious, in which case it will fill `action_index` as it likes, and we can't check anything.
It would help debugging the host impl, but that is easier to do on the host side.

keepkeyjon, 2019-06-14:
> that's sort of pointless, because size is prefilled by your protobuf library.

I should clarify: I mean that the checks enforce the invariant that precisely one of:

*  `data_chunk.size == sizeof(data_chunk.data)`
*  `(data_chunk.size == data_size % sizeof(data_chunk.data)) && isLastChunk`

is true.

>  there's nothing stopping the host from just stuffing 1 MB of data into data_chunk -- and getting back an error because the device doesn't have enough memory.

Isn't it incrementally hashing along the way, and throwing away the intermediate pieces of buffer? If not, it should.

> A wallet needs to know additional info, the maximum allowable size. (messages-eos.options is only needed for the C impl, it's not part of the protobuf definition. i see now that your max size is set to 256; we were planning to assume 2048)

Oh, yeah, great point. Agreed that it's kinda lame that host implementations would have to "just know" that.

keepkeyjon, 2019-06-14:
> Either the host is truthful and sends the requested action_index, in which case it's not telling you much. Or it's doing something nefarious, in which case it will fill action_index as it likes, and we can't check anything.

It would be enforceable on the device side that the `EosActionCommon`'s `action_index` is always monotonically increasing, and that `ActionUnknown` data must all be received before moving on to the next `action_index`. But that doesn't provide much benefit other than when looking at the logs, since the device can enforce those rules even without it.

keepkeyjon, 2019-06-30:
Started on a fix here: https://github.com/keepkey/keepkey-firmware/pull/150

matejcik, 2020-02-27:
After revisiting the problem I'm afraid this is not worth fixing a year after EOS support went live :( Too bad i didn't have time to fix this back then.

closed 2020-02-27 by matejcik