trezor/trezor-firmware #261: Set the threshold screen - redrawing error

sorooris, 2019-06-25:
TT 2.1.2
https://mytrezor.corp.sldev.cz/feature/shamir-backup/development/
Set the threshold screen

Steps to reproduce:
- navigate through the Shamir backup process until you reach the set the threshold screen
- set the threshold over 10
- set the threshold lower than 10

Expected:
The screen is redrawn correctly and does not show any graphical errors.

Actual:
A graphical error is displayed, see the screenshot below.
[image]

andrewkozlik, 2019-06-25:
Same bug appears in _Set number of shares_ screen. Also note the redrawing bug in the words "sharess" and "locationss".

jpochyla, 2019-06-25:
Fixed in 4d0fe4d06404ffa9a3f47ac4eb22ed898a85703e

closed 2019-06-25 by jpochyla