trezor/trezor-firmware #262: Reevaluate touch panel polling

andrewkozlik, 2019-06-25:
It is possible that one of the causes behind the recent issues with the Trezor freezing (https://github.com/trezor/trezor-firmware/issues/144 and https://github.com/trezor/trezor-firmware/pull/217) may have been the way in which we get the touch events from the screen. We use the interrupt polling mode, see https://cdn-shop.adafruit.com/datasheets/FT6x06_AN_public_ver0.1.3.pdf. This means that when a finger is touching the screen we repeatedly send requests over I2C to get the current position of the finger. It amounts to something like over 1000 requests per second. See https://github.com/trezor/trezor-firmware/blob/master/core/embed/trezorhal/touch_t.h#L229. On its own, this doesn't seem to cause any problems, but when it takes place together with another resource intensive operation, such as generating 1000 random numbers every second, it might be causing the freeze issue we observed.

Let's consider some alternative solutions, e.g.:
- Use the touch panel in interrupt trigger mode.
- Use the touch panel in interrupt polling mode, but poll the touch panel no more than every 10 ms.

closed 2022-08-08 by hiviah

bosomt, 2022-08-29:
Is there anything that we can/need do on QA side ?

TychoVrahe, 2022-08-29:
Nothing specific, basically just start using TT builds with this included and if there are any troubles with touch control, please report it

hynek-jina, 2022-08-29:
@TychoVrahe So I'm moving the issue to `Approved`.

PR #2412 (2022-08-08): Optimize touch controller polling
This PR optimizes polling the touch controller by only polling when new data are ready, which significantly reduce cpu load when touch is active. 

Also removed the duplicate data detection, with reasoning that previously we filtered this out because we were literally polling the same data multiple times until new data were ready. We might still receive new data with same values though, not sure we would ever find this useful, so maybe the filtering should stay?

This could be also taken a bit further, using interrupts and DMA we could probably offload most of [... cut]