trezor/trezor-firmware #268: Shamir in progress - Error details: Firmware error

pankrats, 2019-06-27:
Steps to reproduce:
1. having empty device select Recover waller
2. type in the 1 share
3. disconnect your device
4. select Recover wallet again, type in another share (there is no information for user that the 1st share has already been typed)
Result: when you finished typing the 2nd share there is a "Firmware error"

[image]

tsusanka, 2019-06-27:
This is a "leftover" from how things were done before. Since we have decided not to have persistency in the Beta release, we can either:

- instruct the user to enter the following share and do the rest (so it's kind of a persistency already)
- simple show Recovery failed and instruct user to wipe the device

First option shouldn't be that much of work and I suppose it's preferred. Wallet will not currently show any info on that behaviour but I guess that's fine for a Beta.

prusnak, 2019-06-27:
Quick-fix to at least propagate the issue to the wallet: https://gist.github.com/prusnak/877efecb8bb9d7a207b9e39437aaf594

@tsusanka is working on a better fix

tsusanka, 2019-06-27:
Fixed in https://github.com/trezor/trezor-firmware/commit/4f8022c5cdde0416aa2df4d5722d95ab3b37d2a4. Please test again and close if it seems fixed. (I'll send firmware soon)

tsusanka, 2019-06-27:
> Quick-fix to at least propagate the issue to the wallet: gist.github.com/prusnak/877efecb8bb9d7a207b9e39437aaf594

Btw excatly this is what caused https://github.com/trezor/trezor-firmware/issues/278

pankrats, 2019-06-27:
looks like it is fixed (fw version a5d7f2).

Now if I disconnect the device after submitting the 1st share the device is wiped and I need to do everything again (select Create new wallet or Recovery): 

[image]

closed 2019-06-27 by pankrats