trezor/trezor-firmware #270: Shamir Persistence

ZdenekSL, 2019-06-27:


ZdenekSL, 2019-06-27:
Child of #269

tsusanka, 2019-07-09:
Actions items:
- [ ] Persistence starts directly after Recovery confirm page. Words count selection is part of the persistency. That means BIP-39 will be somewhat persistent as well.
- [ ] On homescreen provide info about the progress. If a single share was already entered, we inform how many shares are remaining to reach the threshold. Otherwise we simply show "Enter first/any share" because we do not know the threshold yet.
- [ ] On homescreen offer a button to input share or cancel the process (canceling is equal to wipe - not in dry run!).
- [ ] Send info into Wallet. That this is a Recovery and info about remaining and threshold.

tsusanka, 2019-07-10:
Also let's increase the iteration exponent to 1.

tsusanka, 2019-07-24:
Merged in #347. Still waits for some tests in #342

closed 2019-07-24 by tsusanka