trezor/trezor-firmware #271: Bonus tip about T9 keyboard

ZdenekSL, 2019-06-27:
Child of #269 

Add infobox about T9 keyboard.
Add header option into infobox.

tsusanka, 2019-07-09:
Let's do this. But choose the wording carefully, example: "one by one, but you can speed it up by tapping each button only once (T9 keyboard)". This focues on why and not on what.

ZdenekSL, 2019-07-09:
The text will be decided based on user testing. Matěj is responsible for handling this part, we need just to prepare the header and spacing.

ZdenekSL, 2019-08-05:
Text You can type the letters one by one or use it like a predictive T9 keyboard.

ZdenekSL, 2019-08-20:
Won't be done after discussion

closed 2019-08-20 by ZdenekSL

tsusanka, 2019-09-20:
Removed in https://github.com/trezor/trezor-firmware/commit/cefb1cf4fd9f870da308704565d1ed61461b2c5e#diff-65fbfc4127aa71804b069d407a26dd9fL191