trezor/trezor-firmware #272: Cardano support with Shamir

ZdenekSL, 2019-06-27:
Fix Cardano support with Shamir seed.

tsusanka, 2019-07-02:
@refi93 we are introducing a new form of recovery system, called Shamir backup, defined in [SLIP-0039](https://github.com/satoshilabs/slips/blob/master/slip-0039.md). A kind of replacement for BIP-39. I'm trying to figure out how to support Cardano on top of this, in particular [here](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/cardano/seed.py#L40).

You are calling `bip32.from_mnemonic_cardano`, which converts the BIP-39 mnemonic to entropy and then calls `hdnode_from_seed_cardano`. I'm a bit confused why not call  `bip39.mnemonic_to_seed()` before and insert the seed into `hdnode_from_seed_cardano`. Is there any difference? Is the 
`mnemonic_to_entropy` specific to Cardano?

Thanks!

andrewkozlik, 2019-07-24:
I created SLIP-0023 to solve this issue, see https://github.com/satoshilabs/slips/blob/andrewkozlik/slip-0023/slip-0023.md. In https://forum.cardano.org/t/cardano-with-slip-0039/25443 I asked people to make comments on the new specification in the pull request https://github.com/satoshilabs/slips/pull/712.

I opened a pull request https://github.com/trezor/trezor-firmware/pull/359, but it needs careful testing.

andrewkozlik, 2019-07-25:
I tested #359 on the device to make sure that it gives the same addresses as the SLIP-0023 test vectors. So it should be fine, but testers should also try some Cardano transactions on a Trezor using SLIP-0039 to ensure that everything is in order.

closed 2019-07-26 by tsusanka