trezor/trezor-firmware #274: Refactoring storage.py

ZdenekSL, 2019-06-27:
Child of #269 
Refactoring storage.py

tsusanka, 2019-07-01:
Waits for review in #295

tsusanka, 2019-07-02:
Merged in https://github.com/trezor/trezor-firmware/pull/295

tsusanka, 2019-07-08:
@andrewkozlik as discussed, this is what we need to work with for Shamir backup in storage:

Recovery:
- get identifier of currently entered shares (to warn on incorrect share)
- get indexes of shares that were already entered (to warn on duplicates)
- is recovery in progress?
- get threshold
- get remaining count
- get length of the shares (num of words)

Common:
- get identifier
- get iteration exponent

-------

Questions:
1. Should we store threshold, identifier, remaining and length of shares, if it can be found from the mnemonics that are stored? _I'm voting for yes, because it seems cleaner and it will be deleted when the recovery is finished._

1. How do we store the seperate shares?
  - a) as whole words
  - b) parse and store only the relevant "secret" part

_Whole words are currently easier to implement, but it's not that big of a deal._

1. How to store shares' indexes?
 - a) key = index 
 - b) store shares sequentially and indexes somewhere else (how? bitmap?)
 - c) store shares sequentially, do not store indexes and parse all the shares all the time

_Option a) has the disadvantage that we have to loop through all the possible indexes to find shares. This is low (16) for single-group mnemonics but higher (256) for multiple groups. I'm still slightly in favour of this option, because we can find if some share with some index was already entered very easily._

PR #295 (2019-07-08): core: refactor storage
This PR refactors storage.py. Each namespace now has its own file in apps.common.storage and storage/\_\_init\_\_ serves as an entry point.

Things to be discussed:

- apps.common.storage.device or common or something else?
- Should we prefix the storage.init functions with underscore? They are used by the namespace files, but shouldn't be used outside the storage module.
- apps.common.storage or trezor.storage? As discussed, the difference is quite subtle, so we can keep it as it is and change in case we decide to refactor apps.common vs trezor.

Closes #274.
closed 2019-07-08 by andrewkozlik