trezor/trezor-firmware #276: Complete dry run of all Shamir shares

ZdenekSL, 2019-06-27:
Possibility to run dry run of all Shamir shares. User in the wallet should be asked, after threshold, if he wants to check all remaining shares for validity.

ciny, 2019-07-09:
WIP on this is in ciny/shamir_dryrun branch. UI and workflow should work. Based on discussion at standup we'll finish this up after #270 is done.

tsusanka, 2019-09-03:
@ZdenekSL didn't we decide that we won't do this unless Product decides so?

ZdenekSL, 2019-09-03:
@tsusanka yup, that's why there is product owner label :)

ZdenekSL, 2019-09-05:
Product Owner decided that we will implement this in the future.

sorooris, 2019-09-27:
FYI this issue is also present in the Shamir groups - user is prompted to enter another group shares after an in-group share threshold is reached.