trezor/trezor-firmware #278: Seed partially leaked in Invalid Mnemonic checksum error

pankrats, 2019-06-27:
fw version core-2019-07-bc3a4a

Steps-to-reproduce (i will need to verify it once more):

1. Do Shamir Recovery
2. Enter a share incorrectly (i guess)

[image]

tsusanka, 2019-06-27:
Fixed in https://github.com/trezor/trezor-firmware/commit/c205acd3e186787c789fb1b3fdce0bfe35f08969 . Please test again and close if it seems fixed. (I'll send firmware soon)

pankrats, 2019-06-27:
now, in version a5d7f2, it is fixed. looks good:
[image]

closed 2019-06-27 by pankrats