trezor/trezor-firmware #282: Confusing text "Never make a copy of digital shares..." when creating Single backup

pankrats, 2019-06-27:
Reported by Oliver

Steps to reproduce:
1. An empty device, select Create a New Wallet -> Create with Single Backup
2.  In the new wallet click on Create a backup in 3 minutes

On the device you will see:
[image]

But you instead you will see only one seed displayed (not shares), so the text is confusing.

tsusanka, 2019-06-27:
@bentonoliver so the text should be "Never make a digital copy of your recovery seed and never upload them online!" for BIP-39?

bentonoliver, 2019-06-27:
> @bentonoliver so the text should be "Never make a digital copy of your recovery seed and never upload it online!" for BIP-39?

Yes.

Also, fix the header by changing "Backup" to "Back up".

tsusanka, 2019-06-28:
Fixed in f78e75bc38071e2c650cb0a7526cf529d923bffc.

closed 2019-06-28 by tsusanka