trezor/trezor-firmware #283: Trezor cannot sign this transaction

tuxxy, 2019-06-27:
Hi, we're trying to integrate NuCypher with the trezor to allow staking and signing from it.

We have a transaction that we sign via `ethereum.sign_tx` composed of the following:

`{'value': 0, 'gas_limit': 51988, 'gas_price': 1, 'chain_id': 1337, 'nonce': 37, 'to': b'\xe3\xf0\xa1\xd1oAt1<\xdf0_\xde\x00\xf8\xf9b^\xc1\xab', 'data': b'\xa9\x05\x9c\xbb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf9! \x8d\xeb\x83<\xcb?\xc6\xe1\x80\x1ez,X\x9a\xf3\xe7\xd8\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\tR\xbd\x1d\xc1\xaf\xbc\xa7\xfa\x00\xc0\xaf'}`

When we make this call, it seems the Trezor freezes or something, but there is no output on the trezor nor in the CLI. We're at a bit of a loss as to what is causing this, but I can confirm that this transaction can be signed when it's requested via `trezorctl`.

tuxxy, 2019-06-27:
Here is a test case that demonstrates this hang:

```
from trezorlib import client, tools, ethereum

tx = {'value': 0, 'gas_limit': 51988, 'gas_price': 1, 'chain_id': 1337, 'nonce': 37, 'to': b'\xe3\xf0\xa1\xd1oAt1<\xdf0_\xde\x00\xf8\xf9b^\xc1\xab', 'data': b'\xa9\x05\x9c\xbb\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xf9! \x8d\xeb\x83<\xcb?\xc6\xe1\x80\x1ez,X\x9a\xf3\xe7\xd8\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\tR\xbd\x1d\xc1\xaf\xbc\xa7\xfa\x00\xc0\xaf'}
n = tools.parse_path("m/44'/60'/0'/0/0")
client = client.get_default_client()

ethereum.sign_tx(client, n=n, **tx)```

tuxxy, 2019-06-27:
We're not 100% certain that it's not a bug somewhere before we make this call in our own codebase, but regardless whatever is causing this silent failure is making debugging very difficult or leading us to believe it's somewhere upstream.

tuxxy, 2019-06-27:
I just realized that the `to` address was causing this silent failure because we were casting it to the canonical address.

So, that's what's causing this weird issue. Any ideas why the silent failure occurs?

prusnak, 2019-06-28:
Firmware version?

tuxxy, 2019-07-03:
Unsure, but it's as of this ref from when I compiled it -- `835d5df175c057191e3c9fe2ad31c8546404e854`

prusnak, 2019-07-04:
Try the official latest released version (2.1.1)

tuxxy, 2019-07-09:
I tried it on 2.1.1 and 2.1.2 now to no avail.

matejcik, 2019-07-19:
the problem is that you're submitting the address as bytes. it should be a hex string.

of course, the fact that the failure is silent is still a bug

ZdenekSL, 2019-08-28:
Retest pls @matejcik

matejcik, 2019-08-28:
This is currently fixed in core.

A related problem exists in trezorlib, in that it will silently accept `bytes` in places where only `str` should come in, so I'm leaving this open until that is done.

commit 5b8f5424 (2019-08-29): python: do not accept bytes for str protobuf fields

fixes #283

also adds typing information to misc.py
closed 2019-08-29 by matejcik