trezor/trezor-firmware #286: Recovery: Add a return button to go back when entering words

tsusanka, 2019-06-28:
There is currently no back button when entering words. If I confirm a word and then realize it was wrong, I have no way to go back.

A simple solution is to add a simple backspace button when there is no input in the keyboard, which will take the user one screen back:

[image]

We might need to shorten the text. @bentonoliver could make up the wording? Consider doing https://github.com/trezor/trezor-firmware/issues/285 along with that

tsusanka, 2019-06-28:
@ZdenekSL I'm not sure if we want do this as part of Shamir 2.0 so I'm leaving this up to you (or up to the planning ceremony).

prusnak, 2019-07-02:
Let's do this as part of Shamir 2.0, please.

We can use the backspace button to discard the latest word entered and go one word back.

ZdenekSL, 2019-07-02:
With UX designer, specify the text for a screen if we add backspace.

tsusanka, 2019-07-09:
Discussed with Petr and Matej that this is quite minor issue, which will affect all other users. So we think we should not implement this.

Reasons:
- Users enter the words carefully, so mistakes should occur rarely.
- Persistence will make this easier - only the one specific share needs to be entered again.
- It adds a risk that user will return by mistake.
- It is non-destructive operation if I make a mistake. The worst case scenario you have to enter 20 (or 33) words again (3 min? 5 min?).
- Most of the users will fill in the words correctly with no need for backspace button.

Example: 
Out of 1000 users will do recovery. Out of them 10 people will make a mistake. Because of those 10 people we'd modify the UI (show backspace and shorten the text) for the remaining 990. Those numbers are though a guesstimate.

ZdenekSL, 2019-07-09:
Ok, just for clarification, if I add share, that has a wrong word in it, I can add the same share, with the correct word and it will pass. Correct? If so, please close this issues, we can reopen it if we have user feedback about the need of backspace.

tsusanka, 2019-07-09:
> Ok, just for clarification, if I add share, that has a wrong word in it, I can add the same share, with the correct word and it will pass. Correct?

If you enter a share, which is incorrect (some word wrong / missing etc.) we'll let you know and allow you to enter the share again. Does that answer your question?

ZdenekSL, 2019-07-09:
Yes, thank you.

Hannsek, 2023-09-12:
I quite often enter different word and I have to go through the whole process again. I believe we can crack this either by suggested button or by adding this functionality as a swipe.

Hannsek, 2023-11-17:
**T2B1:** Use [this icon](https://www.figma.com/file/l0gG9XeRJ8FTDQ3cfb1wv9/Firmware?type=design&node-id=8288-87863&mode=design&t=ZiP9jHmp6MdNMlzF-4) with the text `RETURN` instead of `DELETE` button when no symbol is entered. Middle button should be `SELECT`, we can change it everywhere to `SELECT` instead of `CONFIRM`.
[image]

**T2T1:** Let's just do it silently here. So swipe from left to right will trigger the same action as "RETURN" from above.

User will be brought back to the last screen with the letters typed in. See below:
[image]
[image]

matejcik, 2023-11-17:
"return" sounds like "exit the whole process", or maybe "enter" :)
"previous" would be nicer? or "back"

Hannsek, 2023-11-17:
The copy is not final.

b-irving, 2023-11-20:
I think `Previous` works best here.

Hannsek, 2023-11-20:
Thanks, then please use “previous”.

closed 2024-02-19 by mmilata