trezor/trezor-firmware #287: "Never make a digital copy" screen displayed after the backup process

sorooris, 2019-06-28:
TT 2.1.2 a5d7f2

Expected:
User is informed they should never make a digital copy of the shares before they are supposed to make a backup.

Actual:
The "Never make a digital copy" screen is displayed **after** the whole backup process.

tsusanka, 2019-06-28:
This is quite easy to do, but I'm not sure this wasn't intentional. I agree though.

@jpochyla any particular reason why did you put it at the end? Or was it a mistake?

tsusanka, 2019-06-28:
This is a bug. It's quite easy to fix (see tsusanka/note-fix branch), but we need to decide what will be on the final page, because the warning is currently Shamir's last screen:

[image]

@bentonoliver will think up wording, suggestion:

Use the backup
to recover your
wallet if you ever
lose the device.

tsusanka, 2019-07-15:
@bentonoliver so I'm using the text as above?

tsusanka, 2019-07-15:
We now have these two screens, one after another at the end, which might seem cumbersome. We might consider removing the first one, but I'm leaving it as it is for now.

[image] [image]

commit 5134cd0b (2019-07-15): core: add backup warnings at correct place; add backup success page

closes #287
commit 4af4c1bd (2019-08-30): core: add backup warnings at correct place; add backup success page

closes #287
closed 2019-07-15 by tsusanka