trezor/trezor-firmware #288: Recovery: Backspace button is too small

tsusanka, 2019-06-28:
Many consider the backspace button way too small and difficult to press.

Reported by @andrewkozlik

mmahut, 2019-06-29:
I second this, while in recovery I wanted to hit the backspace button while confirmed the bad word. #286 would fix this for my use case too, therefore I think one of these should make it in.

tsusanka, 2019-07-02:
I think this looks quite alright. We should consider doing this for BIP-39 as well, I'll check out what is the widest word there.

[image]

[image] [image]

(downtown is one of the widest words)

prusnak, 2019-07-02:
Can you please try the word "remember"?

tsusanka, 2019-07-02:
The fifth of November! And also mountain, which is quite large.

[image] [image]

prusnak, 2019-07-02:
Looks great! Let's investigate this for BIP39 as you suggest.

andrewkozlik, 2019-07-02:
FYI, the widest word using the Trezor normal font is "downtown" (93 pixels) for SLIP-0039 and "mushroom" (95 pixels) for BIP-0039. But now I realized that we are using the monospace font for the word, So it makes no difference :-D. All eight-letter words are 96 pixels.

tsusanka, 2019-07-02:
For BIP-39:

[image] [image]
[image] [image]

It's all good. I'm also considering moving it one more pixel to the left (but you can barely see it):

[image]

tsusanka, 2019-07-02:
I'm also considering moving it one more pixel to the left (but you can barely see it):

[image]

tsusanka, 2019-07-02:
PR in https://github.com/trezor/trezor-firmware/pull/302

commit f04f190a (2019-07-02): core/ui: enlarge backspace button

closes #288
PR #302 (2019-07-02): core/ui: enlarge backspace button
closes #288
closed 2019-07-02 by tsusanka