trezor/trezor-firmware #290: Shamir recovery: show which identifier were already entered

tsusanka, 2019-06-28:
It would be nice to display which shares (i.e. first four words) were already entered. If someone has 16 shares, it'd be quite useful.

Although this would be nice, I have no idea how to fit it on the screen. We would probably need a button on the "persistance homepage", which would take us to another screen with a scroll page of words?

I'm not sure it isn't too much of an effort. A simpler solution is to just inform the user after entering fourth word, if the share was already entered (see #279), but that of course is not as good, because they do now know which one should they enter.

Reported by @andrewkozlik. Depends on #270.

tsusanka, 2019-07-09:
UX inclines to not implemeting this before data proves otherwise.

tsusanka, 2019-09-03:
@ZdenekSL didn't we decide that we won't do this unless Product decides so?

ZdenekSL, 2019-09-05:
Closed after discussion

closed 2019-09-05 by ZdenekSL