trezor/trezor-firmware #293: Entering the same share several times is taken as valid

pankrats, 2019-07-01:
It seems like an issue for me, if not, please, explain. 

Steps to reproduce:
1. Generate a Shamir backup, for example:
[image]
threshold 8/16 shares

2. Do a Shamir Recovery (20 words). 
3. Enter the same share several times (I entered the 1st share 3 times in a row), result: 

[image]

tsusanka, 2019-07-01:
We currently do not check, if the entered share was already entered or not. Let's discuss on standup if we need to push this into the freeze.

ZdenekSL, 2019-07-02:
We can warn the user after the fourth word was entered.

ciny, 2019-07-09:
I already implemented part of this as part of #276 , it is in ciny/shamir_dryrun branch. As with #276 work on this will be finished after #270 is done. I'll assign this to myself and implement it when persistence is done.

ciny, 2019-07-17:
work on this has been moved to ciny/shamir_word_checks (together with #305 ) and it supports persistence. PR will be made after #270 is merged.

ciny, 2019-07-23:
Code merged from ciny/shamir_word_checks branch to tsusanka/iss270-persistance, it will make it to master as part of #347

tsusanka, 2019-07-24:
Merged to master.

closed 2019-07-24 by tsusanka