trezor/trezor-firmware #296: Device freezes after computer lock

pankrats, 2019-07-01:
It happens with 2.1.2 ver 50f16366

Steps to reproduce:
1. Initialize your device and start "Create new wallet process" (it doesn't matter if you choose Shamir or Simple backup)
2. Lock your computer during the process.
3. Carry on with the process on Trezor device. 

Result: 
at some point, the device will freeze. 
[image]
[image]

It seems not that critical, as if you reload the web wallet the device becomes functional again. 

I will check the 2.1.1 fw version as well.

pankrats, 2019-07-01:
With 2.1.1 the device screen just goes blank during the same use-case. 
So I guess it is not a regression. 
[image]

ZdenekSL, 2019-08-22:
Closed after discussion. Wont be done.

closed 2019-08-22 by ZdenekSL

tsusanka, 2019-08-22:
Because this can't be fixed easily (and is probably more of a bridge issue).