trezor/trezor-firmware #301: Shamir: Refactor mnemonic_bip39 and mnemonic_slip39

tsusanka, 2019-07-02:
They are two seperate files currently, but some of the logic might be merged. For consideration, it might not make sense.

tsusanka, 2019-07-15:
Let's do this on top of #331

tsusanka, 2019-07-16:
Note to the assigne: In #331 I've moved the mnemonic_* files to apps.management.recovery_device folder a renamed them to {bip39,slip39}_keyboard.py. 

I think that was the most important nit-pick. What is left is to consider if we should merge those two keyboards - in other words create some kind of inheritance where we have Keyboard class with common stuff and then Bip39Keyboard and SlipKeyboard as childs. However, I'm really not sure this is desired/needed so let's do it only if it makes sense!

@ciny I think you can take this one, when you're ready.

ciny, 2019-08-13:
@tsusanka I've had a look at this:
Prompt and KeyButton can be moved to some common_keyboard.py easily. With this we should probably use keyboard_*.py pattern.

When it comes to *Keyboard and InputButton - Since we don't have access to the abc module we can't enforce contracts from base classes without doing some contract checks of our own. I had a go at it and it adds a lot of sketchy boilerplate to save about 50 duplicate lines. IMHO it's not worth the potential footgun.

tsusanka, 2019-08-13:
Ok, let's not do it. Maybe it's not worth anything at all then? Moving out Prompt and KeyButton may cause more confusion than good. So maybe let's just do the rename? I agree with that.

ciny, 2019-08-13:
Ok, I'll just rename the files then and submit a PR.

commit 4323b0ac (2019-08-21): core: rename recovery keyboards

closes #301
closed 2019-08-21 by tsusanka