trezor/trezor-firmware #304: Debug build problems for build_unix target

tylerdmace, 2019-07-02:
```Bash
/usr/bin/ld: cannot find -lgcc_s
/usr/bin/ld: cannot find -lc
/usr/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
scons: *** [build/unix/micropython] Error 1
make: *** [Makefile:106: build_unix] Error 2
```

I see reference to this issue in the old `trezor-core` repo and it was closed as being a problem with Ubuntu. I, however, can confirm this issue exists for at least four distributions (clean installs) so I'm inclined to believe that there is a fix somewhere in `SConscript.unix` that resolves library paths better.

matejcik, 2019-07-19:
the problem is that `-no-pie` isn't actually an option that exists in `ld`
we can switch to gcc's `--no-pie`, which maybe does what we want? but i don't know what it is that we want

@prusnak do you know why no-pie is there? if not, i'd drop the whole option. otherwise let's figure out if passing it to gcc instead of ld helps

hiviah, 2019-08-28:
The `DEBUG_BUILD=1` works when you delete `-no-pie` from `SConscript.unix`. Also it seems that you need to enable one ifdef in `py/nlrx64.c` in `vendor/micropython`:

    --- a/py/nlrx64.c
    +++ b/py/nlrx64.c
    @@ -58,9 +58,9 @@ unsigned int nlr_push(nlr_buf_t *nlr) {
         #else
     
         __asm volatile (
    -    #if defined(__APPLE__) || defined(__MACH__)
    +    //#if defined(__APPLE__) || defined(__MACH__)
         "pop    %rbp                \n" // undo function's prelude
    -    #endif
    +    //#endif

Without the nlr patch the emulator segfaults.

prusnak, 2019-08-28:
@tylerdmace What distros have you been trying? This works perfectly for distros I use (including the Debian which is used in the Dockerfile).

hiviah, 2019-08-28:
For some reason, the nlr needs patch for debug build and must not have it without debug build (tried on Ubuntu 16.04 LTS so far)

hiviah, 2020-05-29:
Seems that using `-Og` instead of `-O0` in gcc optimization flags makes the segfault go away without the need of nlr patch.

tsusanka, 2021-10-05:
@hiviah is this issue still relevant?

hiviah, 2021-10-05:
Apparently this still persists in the last three Ubuntu LTS (what I could test). Just invocation changed:

*  `make UNIX_PORT_OPTS="TREZOR_EMULATOR_DEBUGGABLE=1" build_unix`
*  `make UNIX_PORT_OPTS="TREZOR_EMULATOR_DEBUGGABLE=1" PYOPT=0 build_unix_frozen`

I'll fix it by changing the `-O0` to `-Og`. Worked well so far and even the gcc manual seems to suggest `-Og` should be preferred to `-O0` for debugging now.

closed 2021-10-05 by hiviah