trezor/trezor-firmware #309: Help me please Trezor One

sairex28, 2019-07-05:
Good afternoon, I bought a trezor wallet, did everything on the video in YouTube, well, setting the wallet, after clicking on the update to the latest version, I hung up, I updated the page and after that I could not enter the wallet myself, what should I do? Can someone help me? What is the problem? Posted in those. Support is silent for more than 3 days ...

raymonddurk, 2019-07-06:
This isn't the appropriate forum for this. Wipe the device (assuming you have the seed backed up) and download the firmware again.

sairex28, 2019-07-06:
I tried to reinstall, nothing helps ... I do not know where to write with my problem.

sairex28, 2019-07-06:
I wrote "To continue connect Trezor", I insert it, but it does not see it or that I do not understand, I tried different cables already.

prusnak, 2019-07-08:
=> https://trezor.io/support/

closed 2019-07-08 by prusnak