trezor/trezor-firmware #322: Enlarge homescreen image

tsusanka, 2019-07-10:
On model T we currently set the homepage screen as a tiny circle. Marketing mentioned it'd be nicer to have the screen all the way over the display. To be discussed.

prusnak, 2019-07-11:
There is a good reason for that and that is visual identity + we can be sure that the label is always readable. Let's discuss that with UX.

tsusanka, 2019-07-11:
Just to add, this was brought up by Vladimir from marketing (illustrator).

ZdenekSL, 2019-10-15:
IMHO, its because you have the top part reserved for warnings and errors? 
[image]

ZdenekSL, 2019-10-15:
Closed after discussion with the product owner.

tsusanka, 2020-07-03:
Reevaluated, we want do this.

tsusanka, 2021-03-18:
Being done as part of the tt redesign.

closed 2021-03-18 by tsusanka