trezor/trezor-firmware #327: Firmware error - when trying to send Ripple on next wallet

pankrats, 2019-07-12:
Reported via Support ticket.
Unable to reproduce. 

browser type - Google Chrome Version 75.0.3770.100 (Official Build) (64-bit)
OS - Windows 10 version 1803
Trezor Model T
firmware 2.1.1

address we are  trying to send funds to - rwTXopz15WRkXJiD5X2mbojefLZK3RGdYX destination tag 20

Log is attached
[ripple_log.txt](https://github.com/trezor/trezor-firmware/files/3386133/ripple_log.txt)

[image]

pankrats, 2019-07-12:
I sent some ripples to rwTXopz15WRkXJiD5X2mbojefLZK3RGdYX with destination tag 20 and everything went smooth. 

https://xrpscan.com/tx/B6E1CE50EBBF8447FA163737582AE2C2DA020DA18EF30F6384196946CE39D985
[image]

tsusanka, 2019-07-12:
Note to self: try out the amount. 10005849368 is bigger than 2^32 so there might be some overflow or whatnot.

tsusanka, 2019-07-12:
@ciny if you'd like to look at this one, feel free to do so.

tsusanka, 2019-07-15:
We had this [check](https://github.com/trezor/trezor-firmware/blob/tsusanka/iss270-persistence/core/src/apps/ripple/serialize.py#L89) to see if the amount is not higher than the maximum allowed.

However, the maximum as defined in [Ripple's docs](https://xrpl.org/basic-data-types.html#currencies) is most likely in XRP and not in drops ("satoshi" equivalent), which I haven't realized.

Since 100058493680 is higher than the maximum (100000000000), this raised an error. I've fixed this  by modifying the value to the maximum in drops.

tsusanka, 2019-07-15:
Btw: if we would correctly send wire.ProcessError instead of ValueError exception it'd get propageted into Wallet and it'd be easier to debug. I've created an issue for that: https://github.com/trezor/trezor-firmware/issues/332

commit d0731024 (2019-07-15): core/ripple: increase the maximum amount allowed to be sent

The maximum amount is defined in [Ripple's docs](https://xrpl.org/basic-data-types.html#currencies) but it's most likely in XRP and not in drops ("satoshi" equivalent). This commit increase the value to the drops equivalent.

closes #327
commit b122001e (2019-08-30): core/ripple: increase the maximum amount allowed to be sent

The maximum amount is defined in [Ripple's docs](https://xrpl.org/basic-data-types.html#currencies) but it's most likely in XRP and not in drops ("satoshi" equivalent). This commit increase the value to the drops equivalent.

closes #327
closed 2019-07-15 by tsusanka