trezor/trezor-firmware #332: Send wire.ProcessError instead of ValueError where applicable

tsusanka, 2019-07-15:
I've noticed this in Ripple app mostly. For example [here](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/ripple/serialize.py#L90) we should send ProcessError to the user, because it's not an error just some wrong input.

prusnak, 2019-07-15:
Hint: don't put pointers to files in the master branch, they tend to become invalid soon
Hint2: if you are viewing the file in master (or any other branch), press `Y` on a keyboard and github will change the branch name in the URL to actual HEAD of the branch, which you can then use as a permalink

tsusanka, 2019-07-15:
The Refined GitHub extension solves this, so I'm forgetting about it quite often. But you're right of course, fixed.

See:
[image]

tsusanka, 2019-08-14:
Another example: providing invalid Bitcoin bech32 address (see https://github.com/satoshilabs/mytrezor/issues/818) raises Firmware error, but we should communicate to the user that the input is invalid.

PR #874 (2020-02-27): Use wire errors where applicable
Fixes #332.

At most places we have used wire.errors correctly so this was not that big of a deal. There still might be places where wire.error si more appropriate but I did not want to fix something that does not seem broken. So I have fixed the two bugs mentioned in #332, explored the rest and fixed it in places where it seemed reasonable.
closed 2020-02-27 by tsusanka