trezor/trezor-firmware #334: FIDO2 Support txAuthSimple extension

andrewkozlik, 2019-07-15:
The txAuthSimple extension allows for a simple form of transaction authorization. A Relying Party can specify a prompt string, which will be displayed on the Trezor. If the user confirms, then the prompt string will be signed using the credential private key.

Currently I am not aware of any site that would use this, so this is low priority.

We have to be careful about displaying arbitrary prompts on the screen. Since the display is considered trusted by the user, an attacker could misuse this to persuade the user into taking some dangerous action. So we should say something like "Do you wish to authorize the following transaction from <RPID>?". The prompt should have a different color or some other visual distinction. The user will be able to swipe up and down until they reach the end of the prompt and then confirm or decline, similar to the way the recovery seed is displayed.

andrewkozlik, 2021-01-25:
Unfortunately this feature has been removed from WebAuthn, because none of the browsers implemented it. What a shame :-(. This could have been used to securely confirm withdrawals from cryptocurrency exchanges.
https://groups.google.com/a/fidoalliance.org/g/fido-dev/c/nIA0gb3Px6c/m/ag1C9TCCCQAJ?pli=1

closed 2021-01-25 by andrewkozlik

prusnak, 2021-03-04:
The extension was removed via https://github.com/w3c/webauthn/issues/1386

petrdvorak, 2024-03-10:
Help us get it back, @prusnak! https://github.com/w3c/webauthn/pull/2020