trezor/trezor-firmware #338: Add Beam Support

beam-mw, 2019-07-16:
[image]

Hello! The BEAM team is writing to you. BEAM is a next generation scalable, confidential cryptocurrency based on an elegant and innovative [Mimblewimble protocol](https://docs.beam.mw/Mimblewimble.pdf). A lot of work has been done to support our coin in Trezor. We are not in the Top 30 yet, but we are striving for it.

Web Page: https://www.beam.mw/
Block Explorer: https://explorer.beam.mw/
Guide: https://documentation.beam.mw/
Technical Specifications: https://github.com/BeamMW/beam/wiki/Beam-Technical-Specifications
Hardware Wallet Requirements: https://github.com/BeamMW/beam/wiki/HW-wallet-requirements

Not all primitives described in the "Technical Specifications" article are used to implement support on the HW wallet. The following primitives are those that are actually used:

- `secp256k1_gej` - Basic curve point arithmetics: point addition, doubling, negation, import/export to a platform-ndependent format.
- `secp256k1_scalar` - Scalar arithmetics: addition, multiplication, inverse

The following cryptographic functions and schemes are built over them and are used in this implementation:

- Point multiplication (by a scalar). The reason that this functionality is implemented in BEAM and not taken directly from `secp256k1` is the following:
   - We'd like to have more low-level control of the primitives to implement advanced schemes
   - We need more generators: Standard `secp256k1` supports just two (G,H), whereas we need many more (131)
   - No effective aggregation implementation
- Commitments (encoded amount with the blinding factor)
- Schnorr's signatures (including multi-sig)
- Bulletproofs (including multi-sig)

Since the protocol is complicated enough for implementation in a short time, the support of the web (frontend) wallet is not yet considered. A link from the Trezor frontend wallet will lead directly to the desktop version or offer installation instructions.

Thanks for your time, waiting for comments.

Best Regards,
Beam Dev Team

prusnak, 2019-07-16:
Ack, you can start working on a pull request and submit it once it is ready. Thank you!

beam-mw, 2019-07-25:
Hi @prusnak . Pull request was created! Thanks. Beam Team

vonalexbis, 2019-08-17:
So, i can store Beam with Trezor by using my 24 Word seed? Is there a link somewhere? Cannot find it With Trezor. Thanks beamers

prusnak, 2019-08-17:
@vonalexbis not yet

vonalexbis, 2019-08-19:
Thanks for the Info. I thought that I Had read somewhere That it was due for  April 2019 though. Pity. But lets wait and See. Any idea when it will be integrated? Would let me sleep much better :)

tvavrys, 2019-11-22:
ETA? Why is this taking so long?

prusnak, 2019-11-23:
There are things to do with higher priority.

prusnak, 2020-08-07:
Closing, see the reason here: https://github.com/trezor/trezor-firmware/pull/360#issuecomment-668766229

closed 2020-08-07 by prusnak