trezor/trezor-firmware #346: UI for two-level Shamir

andrewkozlik, 2019-07-18:
**Backup**
1. Trezor Wallet or trezorctl will be able to launch one of the following workflows on the device using the ResetDevice message:

    - Single backup,
    - One-level Shamir backup,
    - Two-level Shamir backup.

    It would probably be best to replace the boolean slip39 field in the ResetDevice message with something like backup_type of type int. Note that in the future the ResetDevice message may be enhanced to contain all the information about the thresholds and number of shares in each group.

2. In case of the two-level Shamir backup the user will be asked on the device to enter the number of groups and the group threshold and then to enter the number of shares and the threshold for each group.

3. When listing the shares the device will need to state which group they belong to.

**Recovery**
If during recovery the user enters a Shamir share with group count greater than 1, then the device will need to provide additional guidance to the user during the recovery process, e.g.:

> Completed 1 of 2 groups needed:
> 2 of 2 shares needed from group cargo corner acrobat
> 0 shares from group cargo corner beard
> 1 of 2 shares needed from group cargo corner ceramic

See the CLI in python-shamir-mnemonic for more details.

ciny, 2019-08-15:
- [x] reset device workflow + test
- [ ] recover device workflow + test
- [ ] consider refactoring management/common/layout.py

ciny, 2019-08-27:
done in #428 , polishing and refactoring continued in #442

closed 2019-08-27 by ciny