trezor/trezor-firmware #355: Improve "TREZOR Detected" tooltip condition

szymonlesisz, 2019-07-23:
Rethink conditions for this popup.
Maybe it should be displayed only if device is not initialized?

Invalid use case:
1. Open wallet.trezor.io
2. Connect device (tooltip appears)
3. Click on tooltip (redirection to trezor.io/start)

keepkeyjon, 2019-07-23:
We did different urls based on whether the device was initialized / in bootloader mode:
https://github.com/keepkey/keepkey-firmware/pull/147/files

prusnak, 2019-08-09:
Not showing the popup when a device is not initialized defeats the purpose of the popup completely.

Example:
- malicious person initializes the device and gives it to another person telling them they should go to malicious-trezor-phishing-site.com
- if the popup does not show on an initialized device, the user does not know something is wrong

I agree the UX could be better, but the improvement should be on the trezor.io/start website which should detect whether the device is initialized or not

closed 2019-08-09 by prusnak

tsusanka, 2019-08-22:
@prusnak I'm a bit sad we closed this without any action. I strongly agree with this:

> I agree the UX could be better, but the improvement should be on the trezor.io/start website which should detect whether the device is initialized or not

Should I create a new issue somewhere else?