trezor/trezor-firmware #365: Improve emulator's print screen capability

tsusanka, 2019-07-29:
Currently, when emulator is triggered with `TREZOR_SAVE_SCREEN=1` it makes a print screen. However, it does so on every screen render, which creates a huge amount of useless images (almost duplicates).

We should improve this to check if previous image is not the same (or very similar?) and save only then.

tsusanka, 2019-07-29:
We could also create a seperate tool that would take the images and process them based on their similarity. The user could provide the "level of similarity" they want. For example: 100% will generate all images, 90% the ones without tiny duplicates etc.

prusnak, 2019-07-29:
We already check for the screen contents and don't save a new image in case the new one does not differ from the last one: https://github.com/trezor/trezor-firmware/blob/5584939fd30ac4e0f27a0e28f0f9e5f3a5978f75/core/embed/extmod/modtrezorui/display-unix.h#L230-L237

If that does not work, the feature is broken and needs to be fixed.

@tsusanka are you 100% sure the feature does not work?

prusnak, 2019-07-29:
Also it is trivial to modify the code so that it computes "level of similarity" index instead of 100% match. I'll do that.

tsusanka, 2019-07-29:
I believe that works. But the problem are loaders and other similar screens. In a single reset device tests I get ~50 images with the loader: 
[image] [image]
etc.

Which is makes it quite chaotic. Also in recovery I have a lot of images with the keyboard differing one from another only in the title "Type word XX:".

So we can of course solve this in emulator itself, but I was also thinking about a seperate tool that would compare all those images sequentially and if they are "similar", remove one. You could define the "similarity" index.

Just an idea.

malinah, 2019-11-04:
In post process you can use **ImageMagick** to filter out similar images if that's usable in your workflow.

```bash
# similar images
% magick compare -metric PHASH a.png b.png /dev/null
1.95201%
# NOT similar images
% magick compare -metric PHASH a.png c.png /dev/null
3107.94%
```

tsusanka, 2019-12-18:
I think TREZOR_DISABLE_ANIMATION in #755 fixes this. Let's close and reopen if we deem necessary.

closed 2019-12-18 by tsusanka