trezor/trezor-firmware #375: Shamir testing seed (equivalent of the all seed in single backup)

sorooris, 2019-07-30:
Nice to have for testing purposes and coin management:
- Shamir testing seed, an equivalent of the all seed in single backup

andrewkozlik, 2019-08-28:
Valid Shamir mnemonics of type 1-of-1:

"academic academic academic academic academic academic academic academic academic academic academic academic academic academic academic academic academic rebuild aquatic spew"

"academic academic academic academic academic ocean ocean ocean ocean ocean ocean ocean ocean ocean ocean ocean ocean density email fortune"

"academic academic academic academic academic hybrid hybrid hybrid hybrid hybrid hybrid hybrid hybrid hybrid hybrid hybrid hybrid ajar hour envy"

The first mnemonic might be easier to remember, but the last two are easier to enter, because the words _ocean_ and _hybrid_ can be entered in two key strokes.

To migrate the "all all ... all" seed to Shamir, we would need to support 512-bit shares (59 words), which is possible, but currently we support only 128 bits (20 words) and 256 bits (33 words).

andrewkozlik, 2019-08-28:
Pending https://github.com/trezor/trezor-firmware/issues/455.

tsusanka, 2019-09-18:
@ciny we can close this one?

ciny, 2019-09-18:
I don't see anything more we can do on our side so from my pov we can close this. but that's probably a question for @sorooris since he's the one who reported it so I'll leave closing this to him.

tsusanka, 2019-09-18:
Let's close it here. The https://github.com/trezor/trezor-firmware/issues/442 epic has 1of1 subissue where the testers will test this. cc @sorooris

closed 2019-09-18 by tsusanka

sorooris, 2019-09-18:
ack