trezor/trezor-firmware #384: PIN code change dialogue not properly refreshed

bosomt, 2019-08-04:
Firmware 2.1.3
Minut Linux, Chrome, Stage wallet.

During change of pin try some swiping and touching.
Then press back or confirm and new dialogue is not properly displayed.

I was not able to reproduce it in 2.1.1 but i will try few more times. /it's hard to reproduce it even in 2.1.3
 /  
[image]
[image]

tsusanka, 2019-08-05:
No success in 2.1.1?

sorooris, 2019-08-05:
Same thing occurred while creating a new wallet/backup, while swiping the screen right after confirming the threshold number.

[image]

tsusanka, 2019-08-05:
I've reproduced this in debug mode on device and got this in the console.

```
389526000 trezor.loop ERROR exception:
Traceback (most recent call last):
  File "trezor/loop.py", line 145, in _step
  File "trezor/ui/__init__.py", line 227, in handle_input
  File "trezor/ui/pin.py", line 117, in dispatch
  File "trezor/ui/__init__.py", line 165, in dispatch
  File "trezor/ui/button.py", line 232, in on_touch_end
  File "trezor/ui/pin.py", line 148, in on_confirm
Result: 336050995
389577000 trezor.loop DEBUG finish: <generator object '__iter__' at 200201a0>
389589000 trezor.loop DEBUG finish: <generator object 'request_pin' at 2001f180>
392929000 trezor.loop ERROR exception:
Traceback (most recent call last):
  File "trezor/loop.py", line 145, in _step
  File "trezor/ui/popup.py", line 20, in handle_timeout
Result: None
392947000 trezor.wire DEBUG 0:0 write: <ButtonRequest>
392971000 trezor.wire DEBUG 0:0 read: <class 'ButtonAck'>
392994000 trezor.wire DEBUG 0:0 read: None
```

prusnak, 2019-08-05:
Should be fixed in https://github.com/trezor/trezor-firmware/commit/c9096d9a0a6c1ca7cb34c1c8c205fe5a77d3ea3e

tsusanka, 2019-08-06:
@bosomt @sorooris so we can close this one?

sorooris, 2019-08-06:
@tsusanka couldn't reproduce so i think we can.

bosomt, 2019-08-06:
i was not able to reproduce it anymore after latest fix.
I will close it and if it appers again i will reopen or create new one

closed 2019-08-06 by bosomt