trezor/trezor-firmware #386: Shamir - Sluggish U2F authentization

sorooris, 2019-08-06:
Nixos, Chromium
stage wallet da618940e75c68f9c3b1f41bb1a49fc4ccfb476c
TT 2.1.3 (Shamir, 5edffb32) 

Prerequisites: dropbox or github login with U2F authentization

Steps to reproduce:
1. navigate to dropbox or github and log in
2. wait for the U2F authentization dialogue do be displayed (both on the website and on the device)
3. cancel the dialogue, then confirm it

Expected:
Actions on the device react to the user input with no delays.

Actual:
Compared to 2.1.1 the 2.1.3 cancel and confirm actions have, when logging in with U2F authentization, a slight delay.

2.1.3 here
[image]
2.1.1 here 
[image]

tsusanka, 2019-08-06:
@andrewkozlik could you please check your branch if this occurs on yours as well?

bosomt, 2019-08-06:
works fine for me, i will test dropbox too

bosomt, 2019-08-06:
dropbox works like a charm, with latest signed 2.1.3

andrewkozlik, 2019-08-06:
My branch is working fine. I completely reworked the UI flow for FIDO2 and U2F, so at this moment, there is probably no point in investigating this issue. Once https://github.com/trezor/trezor-firmware/pull/204 is merged into master some time this month, @sorooris please try to reproduce this issue using the same device.

andrewkozlik, 2019-08-30:
I haven't tried to reproduce this, but I am guessing that this issue occurs only when the device is loaded with SLIP-0039 mnemonics. In that case the most likely cause is that we are deriving the seed during every single U2F operation. Every derivation takes about one second. The solution is to cache the seed.

andrewkozlik, 2019-09-01:
I have confirmed that this occurs only when the device is loaded with SLIP-0039 mnemonics and I prepared a simple fix #478. The issue is not very visible during registration, but mainly during authentication. After seeing the issue live, I feel that it is more pronounced than I thought before. We should consider the possibility of including the fix in the upcoming release of Shamir firmware.

sorooris, 2019-09-02:
Retested on 2.1.5 c529875 with 2/2 Shamir seed; it's OK and seamless.
Will retest on 5/8 just to be sure.

sorooris, 2019-09-02:
Retest on 2.1.5 c529875 with 3/5 Shamir seed OK (Passphrase on/off, PIN on/off).
Retest on 2.1.5 c529875 with 5/8 Shamir seed OK (Passphrase on/off, PIN on/off).

closed 2019-09-02 by sorooris