trezor/trezor-firmware #390: Shamir UI feedback

prusnak, 2019-08-07:
With the mnemonic on device recovery for Trezor T, the top left [<-] delete box is far too small, and the green word confirm button is far too big. It is very difficult to delete without instead clicking the green confirm word box. Perhaps the best solution is to add a dead non-click spacer next to the [<-] button, so the green confirm is not accidently clicked instead.
It would also be good if you could go backwards to the precious word by clicking the [<-] delete button.

tsusanka, 2019-08-07:
Still happens to someone eventhough we have enlarged the backspace? See https://github.com/trezor/trezor-firmware/issues/288

Could you verify this concerns 2.1.3? Because it is obsolete otherwise

prusnak, 2019-08-07:
Duplicate of https://github.com/trezor/trezor-firmware/issues/288

closed 2019-08-07 by prusnak