trezor/trezor-firmware #395: Weird UX for Shamir 256-bit shares

prusnak, 2019-08-09:
Firmware 2.1.3

Steps to reproduce:

```
$ trezorctl reset-device -e -t 256 -b shamir
```

[image]

[image]

I would expect that words 31+32+33 will be shown on the same page.

tsusanka, 2019-08-13:
Option 1:
[image] [image]

Option 2:
[image]

Option 3:
[image]

prusnak, 2019-08-13:
Option 1 seems the best.

tsusanka, 2019-08-14:
I've chosen the first option then.

tsusanka, 2019-08-14:
I've also opened https://github.com/trezor/trezor-firmware/issues/409

PR #408 (2019-08-14): core/slip39: fix UI for 33 words
Please review. I'm a bit scared doing anything in SLIP-39 without review :).

closes #395
commit e119e8de (2019-08-14): core/slip39: fix UI for 33 words

closes #395
closed 2019-08-14 by tsusanka