trezor/trezor-firmware #400: Shamir Backups are completely broken

Matt-OG, 2019-08-11:
Using my new Trezor T (firmware 2.1.3) and beta-wallet.trezor.io I wiped and used "create a wallet with Shamir Backup".

-----

1st attempt : 
I did the backup, chose 3of5 and recorded the 20 words fragments

I recorded the ypub and xpub in the Basic tab, then in Advanced > Check recovery seed.

I entered the fragments #5, #3, #4.

> X Warning
> The entered recovery shares are valid but do not match what is currently in the device

I then wiped the Trezor T in beta-wallet and then "Recover your wallet" > 20 words.

Entering fragments #5, #3, #4 the ypub and xpub were different to those shown in beta-wallet before the wipe.

I then wiped again, and restored #1, #2, #3 with the same ypub/xpub as the #5, #3, #4 restoration.

-----
2nd attempt:

I wiped again, and in beta-wallet "Create a wallet with Shamir Backup" and recorded the 3of5 20-word-shares.

I recorded the ypub and xpub in beta-wallet

Then wiped Trezor

Then recovered fragments #3, #4, #52 

beta-wallet assigned this restore a completely different device ID URL (and as i didn't forget the pre-wipe device) it created another trezor in the sidebar with a different ID, ypub and xpub.

---

I have not yet tried restoring a `trezorctl reset-device -t 256 -b shamir` seed yet to see if the bug is the same.

prusnak, 2019-08-11:
Having a different Device ID after the wipe is expected. The Device ID is regenerated on every device initialization.

The rest is not expected, though.

prusnak, 2019-08-11:
@sorooris Can you please retest 2.1.3?

tsusanka, 2019-08-11:
I'm afraid this is genuine and is caused by the fact that the `slip39.generate_mnemonics_from_data()`;` function takes `master_secret` as an argument instead of `encrypted_master_secret`. We will investigate properly tomorrow.

sorooris, 2019-08-12:
> @sorooris Can you please retest 2.1.3 with the above two scenarios?

1st scenario replicated successfully, will test the second one.

tsusanka, 2019-08-12:
Fixed via #403. We have also released a hotfix fw 2.1.4 now avialable on Trezor Beta Wallet.

Thank you very much for your report!

PR #403 (2019-08-12): shamir hotfix EMS vs MS
identical to #402, except based on a release branch

fixes #400.

DRAFT for now because it points to the release branch. Before merging, we must decide whether to merge there, or create a hotfix sub-branch.
closed 2019-08-12 by tsusanka