trezor/trezor-firmware #405: Monero tests are sometimes failing

tsusanka, 2019-08-12:
Monero tests sometimes fail on CI with no proper error message, but pass on rerun, which is really annoying.

tsusanka, 2019-08-14:
Example: https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/272078687

tsusanka, 2019-08-14:
This happens in about 30% of the Job runs. @ph4r05 do you think we could improve what the tests print out? In other words, any ideas how to debug this?

ph4r05, 2019-08-14:
Thanks for the report. I will take a look.

tsusanka, 2019-10-04:
I've added `retry` parameter in 737ebeec1c8ba91c7d5d78c598a6cabd7e4785e8 to the appropriate job in the CI as a hotfix so it should not cause problems on the CI at the moment. However, this should be fixed properly not to use the CI's power unnecessarily.

ph4r05, 2019-11-08:
Update: I've finally managed to get HF12 (the newest hard fork planned) tests working. All tests passed so we should really be ready for HF12.

I will publish a new testing binary soon. We will see whether it resolves also this problem.

tsusanka, 2019-11-13:
I have bad news :-/ https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/349612540

I'm wondering if it is not related to the fact that the binary is cached in gitlab.

tsusanka, 2019-11-13:
@prusnak had a good point that we have not rebuilt our docker environment. Triggered now let's how see what happens

ph4r05, 2019-11-13:
Hm it is strange that there is no log printed out. We will need it for further investigation.

prusnak, 2019-11-13:
@ph4r05 this was the previous version (0.14.x) used in the job above. False alarm :)

tsusanka, 2019-11-13:
Indeed, my apologies. On new environment (that means 0.15.x) it [passes](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/349673531), so I'm closing again.

I'll also modify the script, which runs the test to print the Monero binary hash so we know what is actually running.

Sorry for the confusion.

closed 2019-11-13 by tsusanka

ph4r05, 2019-11-13:
@tsusanka feel free to update binary to the

- `https://github.com/ph4r05/monero/releases/download/v0.15.0.0-tests-u18.04-02/trezor_tests`
- `36852ff2add3f865dbf4e23f4e32e1022a8f4b14cd4fb4eb97491d99ba93b9bc`
- Release info: https://github.com/ph4r05/monero/releases/tag/v0.15.0.0-tests-u18.04-02

This updated version better handles Trezor lookup errors. If the problem with tests failing repeats we may also need the log file produced by the testing binary.

prusnak, 2019-11-13:
@ph4r05 Added in 0a0cd797e38838dcb0e23f230653094863923517 - let's see in the CIO

PR #699 (2019-11-12): WIP tests: use new monero testing binary v0.15.0.0
- tests new HF12
- attempt to fix #405