trezor/trezor-firmware #406: Split Dockerfile to build and testing versions

tsusanka, 2019-08-13:
Since we are signing firmware built using the ci/Dockerfile it'd be better to seperate test environment and build environment. We can either have multiple Dockerfiles where one extends the other or use variables.

tsusanka, 2019-08-27:
We want to do this before any release, meaning we need to do this till September 4th.

tsusanka, 2019-08-27:
@prusnak I have two questions regarding this:

1. If we remove things needed for tests, the tests will obviously fail on this image. That is fine, because we will use that image for building exclusively, correct?
2. I'm thinking about putting back `debian` instead of `python` docker image, because we have number of Python versions anyway. Should we use Debian 10, which is the latest stable or rather 9?

prusnak, 2019-08-27:
1. correct
2. `debian:10` sounds nice

prusnak, 2019-08-27:
Fixed in 82fdf7e03d527e6294da4065b31e41257d015c9a

closed 2019-08-27 by prusnak